vortexntnu/vortex-auv #53: Restructure codebase

michoy, 2020-09-13:
The directory structure can be confusing, and should be restructured into a more intuitive format. A new format should be decided on, and a first proposition is provided below:

**Motion**: thruster control and path following 
**Navigation**: Localization, mapping and path planning 
**Mission**: State machine and states 
**Arm**: Controller for arm 
**Object Detection**: Detection of objects visually and with acoustics

michoy, 2020-09-14:
Update:
~~**Arm**~~
**Manipulators**: Arm and torpedo

michoy, 2020-09-16:
Additional folder: 
**Config**: Files for robot describtion

chrstrom, 2020-10-02:
The current work on this issue can be found [here](https://github.com/vortexntnu/Vortex-AUV/tree/improvement/new_file_structure)

chrstrom, 2020-10-17:
> The current work on this issue can be found [on this branch](https://github.com/vortexntnu/Vortex-AUV/tree/improvement/new_file_structure).

This work has now been merged with [development](https://github.com/vortexntnu/Vortex-AUV/tree/development/)

chrstrom, 2020-10-18:
The restructure now also includes the restructuring of launchfiles B) Since there are a few crossdeps with the simulator, this part might be a bit messy, but hopefully it will be all okay in the end

closed 2020-11-04 by michoy