vortexntnu/vortex-auv #54: Set up CI for build tests

michoy, 2020-09-13:
Github Actions can be used to set up CI. There is an action on the Actions marketplace specifically for ROS environment setup, so it might be doable without too much work

theBadMusician, 2020-11-18:
Implemented [a simple CI action](https://github.com/vortexntnu/Vortex-AUV/blob/feature/dev_ops-CI/.github/workflows/ROS_build.yml) to [a new branch](https://github.com/vortexntnu/Vortex-AUV/tree/feature/dev_ops-CI) which gets all ROS dependencies, sets up a catkin workspace, clones other Vortex repos, builds all packages, and finally runs a finite state machine on the Vortex Simulator with Vortex AUV modules.

The action starts when there is a push to [the CI branch](https://github.com/vortexntnu/Vortex-AUV/tree/feature/dev_ops-CI). It takes around 15 minutes to complete because [it needs to get the full-desktop ROS Melodic distro and build all of the Vortex related packages](https://github.com/vortexntnu/Vortex-AUV/runs/1386190140?check_suite_focus=true).

Currently it is a really elementary test sequence, as it declares the Github action successful as long as everything builds and runs without fatal errors. Further down the line, it would nice to implement a test-run which requires that the simulated AUV completes a dedicated task e.g. goes through a gate as predicted.

chrstrom, 2021-03-02:
Closing this now, as we will get a more thorough introduction to Docker soon, and should probably make more specific issues when implementing this in our pipeline (if we can even call what we have a pipeline lol)

closed 2021-03-02 by chrstrom