vortexntnu/vortex-auv #74: Gather camera footage for testing purposes

detoxte, 2020-09-27:
The object detection, stereo camera, and ORB-SLAM system requires testing data from an underwater camera rig to be tested properly. This footage needs to be gathered.

chrstrom, 2020-09-30:
Update: There exists a camera rig at ITK that we should be able to use to gather this footage. More info about that to come.

michoy, 2020-09-30:
The rig was created by Andreas Våge a few years ago. He probably knows whereit is. It has to flir cameras mounted in a stereo config. Anette Stahl is a Professor at the institute who works with navigation using vision, she should also know about the rig.

michoy, 2021-01-31:
Camera footage should be recorded from the drone at pool test 2

chrstrom, 2021-03-02:
Pool test 2 here, when do you plan to do this?

prinsWindy, 2021-03-02:
Me and @torgeirgrini will most likely be available in the afternoon to do this. Does that work with Control's schedule?

closed 2021-03-19 by prinsWindy