vortexntnu/vortex-auv #79: Report for previous work on acoustics

chrstrom, 2020-09-27:
A report for the previous work on acoustics needs to be written before this year's work on the acoustics.

The report should contain:
- What has been done in the previous year
- What worked or went well?
- What did not work or went badly?
- Any other valuable experiences from the previous work
- What needs to be done this year

chrstrom, 2020-09-27:
Attached is the input from Henrik, who worked with acoustics last year.
[Akustikk-input-fra-Henrik.docx](https://github.com/vortexntnu/Vortex-AUV/files/5288594/Akustikk-input-fra-Henrik.docx)

oyssolbo, 2020-11-05:
[Report acoustics.txt](https://github.com/vortexntnu/Vortex-AUV/files/5496368/Report.acoustics.txt)
Small update for the acoustics. Should be updated by someone on the HW-team

closed 2020-11-05 by oyssolbo