vortexntnu/vortex-auv #85: Test all hydrophones - one might be broken

michoy, 2020-10-06:
One of the hydrophones diverged from the other two last year, so we should check if it is broken. This is important, as one hydrphone costs ~10k NOK

theBadMusician, 2020-10-18:
The newer hydrophones were tested in the test pool today, powered with 12V from an electric DC generator. 

Two out the three hydrophones functioned without problems, generating a peak voltage of around 5V during pinger signal. The last hydrophone had a problem, as it didn't indicate anything at first. It was moved around a bit, after which it detected pings for about 3-5 seconds and then stopped.

One possible cause for the issue could be a broken connection between wires in the hydrophone cable, but it's unclear at this point. Further testing of the hydrophone/cable is required.

oyssolbo, 2020-12-18:
> The newer hydrophones were tested in the test pool today, powered with 12V from an electric DC generator.
>
> Two out the three hydrophones functioned without problems, generating a peak voltage of [...]

Do you know if anymore testing has been done, Benjamin?

theBadMusician, 2020-12-19:
There has been conducted one more test of the hydrophones by Alexander Øvergård. IIRC he said that the defective hydrophone behaved differently (better) than described in [my comment above](https://github.com/vortexntnu/Vortex-AUV/issues/85#issuecomment-711174823), but you should ask him for details.

oyssolbo, 2021-01-16:
Closing the issue. See discussion on teams/acoustics taskforce

closed 2021-01-16 by oyssolbo