vortexntnu/vortex-auv #135: Communication between Xavier/ROS and the MCU

chrstrom, 2021-01-31:
* Create an overview of how the MCU_interface node for manta works. (Setup a meeting with Øystein and Arthur)
* Figure out what communication protocol that should be used (SPI, I2C, UART etc).
    * Implement the protocol on the MCU side
    * Implement the protocol on the ROS side
    * Replace crc with standard/lib failsafe code
* Testing: Setup on the Xavier. Make sure communication works as intended.

heiring, 2021-02-08:
Using https://github.com/amaork/libi2c (MIT license), we've set up I2C-communication between the Xavier and the MCU. Currently we're only able to transmit strings. What needs to be done:
* The messages received on the /pwm-topic, which the mcu interface subscribes to, are of type pwm.msg, which contains two arrays of type uint16. We need to figure out how to transmit these arrays over the I2C-connection. 
* Create a ROS-node which subscribes to /pwm, includes libi2c an transmits the pwm-messages to the pwm.

heiring, 2021-02-14:
I've created a new branch called feature/xavier-mcu_interface and set up a ROS-node for the new mcu interface.

chrstrom, 2021-02-14:
I think it is smart to put everything in a separate package for now. We should, however, come back to this after everything is functioning on the new drone, and structure everything appropriately.

Btw, is this branch stale https://github.com/vortexntnu/Vortex-AUV/tree/feature/mcu_development now that feature/xavier-mcu_interface exists?

heiring, 2021-02-14:
Not familiar with that branch, looks pretty stale to me.

chrstrom, 2021-02-15:
> 
> 
> Not familiar with that branch, looks pretty stale to me.

Went ahead and removed it, since it didn't have any commits to it 😄

chrstrom, 2021-02-17:
The branch https://github.com/vortexntnu/Vortex-AUV/tree/feature/xavier-mcu_interface now contains an outline for the mcu_interface. The function [transfer_to_mcu](https://github.com/vortexntnu/Vortex-AUV/blob/06f15ecd4de2beaf1099b7a6bb3871655c0e9360/motion/mcu_interface_new/src/mcu_interface.cpp#L98) is the main part that's left to do.

chrstrom, 2021-02-25:
This is mostly fixed, but a cleanup and documentation is needed before merging with development :smile:

chrstrom, 2021-02-25:
Closing as it has been implemented, and this should be reflected in the milestone for pooltest 2. Not closing with PR for the above reasons

closed 2021-02-25 by chrstrom