vortexntnu/vortex-auv #137: Calibrate camera under water

michoy, 2021-01-31:
- [ ] Understand the process 
- [ ] To a trial over water
- [ ] Calibrate under water at the next pool test

michoy, 2021-02-14:
The calibration board is missing, might be at mc-lab

michoy, 2021-02-26:
@prinsWindy do you have time to test the calibration procedure with the board sometime today?

prinsWindy, 2021-02-26:
@michoy I am unavailable today, but will be available for testing tomorrow.

prinsWindy, 2021-03-21:
The camera calibration footage has been collected. However, stereo calibration with [Kalibr](https://github.com/ethz-asl/kalibr) isn't working (error: Optimization diverged). 

As I haven't gotten any answers from Kalibr, I will test different calibration softwares such as OpenCV. If this still doesn't work, then the calibration footage must be inadequate and should be recollected during the next pool test.

michoy, 2021-03-31:
@prinsWindy where are we on this?

prinsWindy, 2021-03-31:
My best guess is that the calibration footage is bad (too rapid movements probably). In addition the calibration suffered from some bandwidth limitations, as there was a difference in the amount of images stored from each sensor. 

I definitely think that new calibration footage should be gathered. Something should be done about the bandwidth issue: lower the framerate, look into the performance of the PC that rosbags the footage etc. The calibration footage should be gathered with someone in the water with the drone, so that the movement can be controlled better.

chrstrom, 2021-07-16:
No new info here, but this will become part of the next year's work for object detection. Closing and noting down a task for them :))

closed 2021-07-16 by chrstrom