vortexntnu/vortex-auv #140: Set up localization

michoy, 2021-01-31:
Plan A) Use ESKF

- [ ] Add to Navigaion folder
- [ ] Get it running on Xavier
- [ ] Refactor code - delete unused lines and improve naming here and there (talk with Øyvind)

Plan B) Use robot localization
- [ ] Get it running on Xavier
- [ ] Tune it

torgeirgrini, 2021-02-08:
Cloned the ESKF repo from https://github.com/vortexntnu/ESKF into the navigation folder, but failing to build the package using catkin build.
[image]
[image]
[image]
If anyone have any suggestions regarding building the eskf package, please let me know.

torgeirgrini, 2021-02-14:
To build the ESKF package, it is necessary to use
#include <vector>
in the ESKF.h file

michoy, 2021-02-14:
#149 should be fixed together with this

michoy, 2021-02-26:
@prinsWindy  @torgeirgrini could you summarize where we are on this?

torgeirgrini, 2021-03-07:
Nearly all of the parameters has been set in the ESKF for gladlaks. The missing parameter is dvl alignment, which requires that issue #189 is solved first.

closed 2021-03-14 by torgeirgrini