vortexntnu/vortex-auv #145: joystick node guidance by wire

jae-hyeong-hwang, 2021-02-07:
Joystick node needs to be published to setpoints for the reference model.

* open loop
* depth hold
* pose hold
* depth heading hold

that modes need to be bounded for control buttons.

chrstrom, 2021-02-07:
Make sure to keep integral action always off for the dp controller

chrstrom, 2021-02-09:
Here is the proposed structure for solving this issue.

[image]

chrstrom, 2021-02-15:
Pull #152 implements the structural changes to the joystick system, minus the control mode select. (This will come when the joystick is to be integrated with the controller

oyssolbo, 2021-03-09:
Should have that we can use the "start" on joystick to switch between ROV and AUV-state or something

jae-hyeong-hwang, 2021-03-09:
> Should have that we can use the "start" on joystick to switch between ROV and AUV-state or something

remember this talk with you on sunday. In other words, "start" to make on and off the control by joystick?

chrstrom, 2021-03-11:
> 
> 
> Start in AUV-mode by default, but possible to switch to ROV-mode via the joystick. I have written some code, but not published as of yet

Especially for Tau, I think having the ability to intervene at any point is good. We might have to add something in the current state machine that exists whatever it is currently doing when the joystick is activated? For the potential robosub stuff, we won't have a tether/joystick connected, so it seems safe to have it this way 👍

chrstrom, 2021-03-25:
Closing this now that #162 has been merged

closed 2021-03-25 by chrstrom