vortexntnu/vortex-auv #146: Investigate possibility of static IP addresses for office computers

prinsWindy, 2021-02-07:
To work on the office computers (Xavier, vortex-hvit etc.) from home, we may SSH into them. 

If any of these computers are rebooted, they may change IP address as they are dynamic. We should contact NTNU (perhaps the IT department) to find out whether or not we may set static IP addresses for our devices.

chrstrom, 2021-02-11:
Static ip addresses can be set in the /etc/network/interfaces file. The only thing I am not sure about is whether or not the DHCP server knows that these IPs are fixed by us. If not, we might end up with conflicting addresses. Having a reserved space would potentially be something we can bring up with whoever is responsible for IT

chrstrom, 2021-02-15:
Closing this as Michael has contacted IT, which should resolve this :smile: We will hopefully get an ethernet wall plug by the switch that actually holds the cable in place

closed 2021-02-15 by chrstrom