vortexntnu/vortex-auv #155: only launch one thruster interface

michoy, 2021-02-17:
**Problem**
today both thruster interface and mcu interface are launched ona "real" launch. Both do the same job, but mcu interface is for thruster connected to a mcu and thruster interface is for thruster connected to an adafruit board on an raspberry pi. Funny thing is that Manta after its refit now is using the setup that thruster interface was made for. 

**Solution**
 Seperate the two nodes in launch, such that the correct node is launched depending on what drone is used.

chrstrom, 2021-02-17:
The thruster_interface node currently does something that is needed for the gladlaks mcu; namely converting a thruster_forces vector into PWM signals. I discussed with Håvard and the best solution seems to let the mcu_interface take care of that conversion, and also send the pwm message to the mcu. This removes the need for a separate topic for the pwm message. 

Since we are approaching the integration test so fast, the mcu_interface will most likely be designed for the gladlaks first. Once we have more time, we want to eliminate the thruster_interface entirely, and make the mcu_interface usable for any hardware. The hardware specifics will be put in the configuration files.

michoy, 2021-02-17:
Thanks for the clarification - seems I got tricked by our own documentation.

The move to include the pwm conversion on the mcu interface sounds very reasonable to me as well.

chrstrom, 2021-02-25:
Note: This is currently fixed, but with hardware-dependent files in the repo. I believe it could be a better idea to add everything to a central MCU repo, but this will likely happen after the pooltest

chrstrom, 2021-03-02:
Closing as this has now been resolved

closed 2021-03-02 by chrstrom