vortexntnu/vortex-auv #156: Xavier acting weird

ingar-eh, 2021-02-17:
The Xavier has 3 problems right now;

(1) Xavier automatically reboots every ~20 minutes, sometimes less.
(2) Fan runs continuously at max speed. You can overwrite manually by editing the value in target_pwm (0 is max, 255 is off).
(3) USB and Ethernet-port doesn't work

The source of problem (3) might be linked to the SPI / I2C-configuration done a week ago. The source of problem (1) and (2) I have no clue.

chrstrom, 2021-02-17:
As a note, the ethernet port works fine, it just needs to be configured. For now the USB-C ethernet can be used.

michoy, 2021-02-17:
Have you checked the performance settings? You can set the Xavier to run at several different performance levels. At the lowest ones, the fan should not need to run. If you set it to a high one, the fan will run continuously. 

The rebooting is definitely something we need to figure out. My first guess is that it might be due to overheating. We should also check what processes are running in the background. Maybe there is something power-demanding running that should not be there.

michoy, 2021-02-17:
Regarding problem 3: I remember someone saying that enabling SPI / I2C had disabled the USB port

ingar-eh, 2021-02-17:
> Have you checked the performance settings? You can set the Xavier to run at several different performance levels. At the lowest ones, the fan should not need to run. If you set it to a high one, the [...]

I checked to make sure that the CPU wasn't running at constant 100% capacity, it wasn't, which makes sense because the fan is blowing cold air. I ran a diagnostic which claimed the fan was running in "Quiet mode" and power consumption was at "Max". 

Maybe Xavier thinks it's overheating which leads to max fan speed and a forced shut down. We could check the temperature gauges and simulate some artificial temperature nodes to see how the Xavier responds to different heat levels.

michoy, 2021-02-17:
just to be 100% sure - you checked the performance menu in the top right corner?

ingar-eh, 2021-02-17:
> just to be 100% sure - you checked the performance menu in the top right corner?

No, I used a series of terminal commands from ssh. I don't remember the exact commands from memory. Are you referring to the Xavier / Ubuntu GUI when you say top right corner? Interacting with the GUI is hard without a functioning USB-port.

michoy, 2021-02-17:
Yes, that's what I was referring to. I can see how the lack of USB might be an issue there. If the USB-c ports are still functioning, we can connect a mouse and a keyboard through adapters. That might simplify the debugging a bit.

ingar-eh, 2021-02-18:
[image]

michoy, 2021-02-26:
Update: The random restarts seem to have disappeared, but the fans PWM signal is probably inverted (it is at max when it should be quiet and vice versa)

chrstrom, 2021-03-02:
Issue #179 addresses the fan/temperature problems. The rest of the Xavier bugs seem to have gone away though.......... Closing this for now, but I think it will be a good idea to reflash the Xavier sometime soon.

closed 2021-03-02 by chrstrom