vortexntnu/vortex-auv #160: Clarify what is sent between sensors and localisation

michoy, 2021-02-20:
The topic names, message types, and contents of messages is not intuitive enough today. This should be refactored such that names, topics, and contents of messages are self-explanatory.

michoy, 2021-03-14:
@torgeirgrini any updates on this?

torgeirgrini, 2021-04-11:
The ESKF takes inn two nav_msgs/Odometry, one which holds the z value in meters from the pressure measurement, and one which holds the linear acceleration values from the DVL (pressureZ_topic and dvl_topic in parameters yaml-file). These values are all published in one topic from the DVL (auv/odom), https://github.com/vortexntnu/dvl1000_ros. The ESKF also listens to a topic from the IMU (imu/data_raw) https://github.com/vortexntnu/stim300.

chrstrom, 2021-08-15:
Closing in favor of the new work on the EKF

closed 2021-08-15 by chrstrom