vortexntnu/vortex-auv #163: Add Battery monitor

Areskiko, 2021-02-21:
The batteries can become damaged if their voltages drop too low. A ros-node that informs and warns of the levels should be added. Additionally it should stop the drone if the voltages reach a critical level

chrstrom, 2021-02-21:
[These docs](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/power_management_jetson_xavier.html) should give you everything you need in terms of monitoring functionality for the Xavier. If the voltage dips below 14V, this ROS node should spam the console with a ROS_FATAL message, before killing the entire Xavier. Since the drone is positively buoyant, it will float to the surface after this.

This approach leaves the DVL and sonar still running. If the electronics group is nice enough, we will be able to control the killswitch from a GPIO port on the Xavier.

Areskiko, 2021-02-22:
Where should the monitor files be located, should it have it's own top-level folder? It doesn't seem to fit in any of the other ones

michoy, 2021-02-22:
I think it makes sense to out it in Mission, as the monitor is very relevant to the abortion of a mission

Areskiko, 2021-02-23:
If the electricity gets transformed before it's delivered to the drone, it could obscure the voltages of the batteries. I've asked Electrical for aid in remedying this.

Areskiko, 2021-02-23:
I've talked to them, and apparently the electricity doesn't get converted, this resolves the problem

Areskiko, 2021-02-23:
I have checked the commands on the Xavier, the voltages concur with the power cable, though power and current seem to deviate. The values are provided in plaintext with no units or text attached, it is worth to mention that the voltages are provided in millivolt

PR #169 (2021-02-25): Battery monitor
Creates a node that publishes battery voltages, as well as using `rospy.loginfo`, `rospy.logwarn`, and `rospy.logfatal` as appropriate

closes #163
closed 2021-02-25 by Areskiko