vortexntnu/vortex-auv #164: Gladlaks parameters

chrstrom, 2021-02-23:
The gladlaks.yaml file needs to be filled with parameters. [(Here)](https://github.com/vortexntnu/Vortex-AUV/blob/feature/gladlaks_parameters/auv_setup/config/robots/gladlaks.yaml)

Note that the thruster placement will most likely change at the start of the pooltest after mechanical is done with their stuff (!!).

chrstrom, 2021-02-25:
If everything goes according to plan, this will be done by saturday/sunday morning :smile:

michoy, 2021-02-26:
Update: there are several parameters in Manta.yaml that are no longer in use. Some of them have already been removed in galdlaks.yaml, but there is still some cleanup left todo.

aksekris, 2021-02-26:
For what I'm working on I'm scrapping almost all the old parameters and only adding the ones I need. So I think I have a pretty good overview of what the simulator needs and not.

chrstrom, 2021-03-05:
Closing this as we have a pretty clear idea of the parameters now; any adjustments from hardware can be reflected in the parameters relatively quickly

closed 2021-03-05 by chrstrom