vortexntnu/vortex-auv #167: dae file of gladlaks are badly scaled

aksekris, 2021-02-23:
Ingar, do you suggest we make a new one or just scale the current one down a couple of notches? Is it a lot of work to make a new one? Looks like there is an option to scale the current one (in the xacro file), but I haven't got to try it yet since manually scaling it will take an iterative process to get it right.

ingar-eh, 2021-02-23:
The scale I have selected for the gladlaks-model in the .xacro right now is arbitrary and not exact. If we want it to be exact we can:
1) Do some calculations / measurements and use the scale parameter in the .xacro 
or 2) Export using an exporter that supports dimensions. .dae-files are basically XML-files, and if you look in the manta_shell.dae you will see a paramater "dimension = centimeter" (or something similar). This is because manta_shell.dae was exported using Assimp which formats the .dae files differently from the one I used on gladlaks.dae (meshlab).

However, to my knowledge, the .dae files (both for gladlaks and the thrusters) are purely visual, the actual collision geometry of the AUV is a sphere with a center in COM, and the thrusterlayout used in the model shouldn't affect the controller. If we do want it to be perfectly exact I suggest exporting the .dae anew using a exporter that supports dimensions.

aksekris, 2021-02-23:
Alright, nice point about the collision geometry. Perfection is probably not worth it so I will try to do some scaling first. Apart from the size the model looked sick btw :)

aksekris, 2021-02-24:
Scaling it x0.01 in xacro file worked out great!

closed 2021-02-24 by aksekris