vortexntnu/vortex-auv #189: IP change for DVL

michoy, 2021-03-03:
Since the Xaviers static IP had to be changed to match the mc-labs subnet, the DVL IP now also needs to be adjusted. This can be done by a windows software supplied by Nortek. We do not know exactly where to find that program yet, but this [pdf](https://www.nortekgroup.com/assets/software/N3015-006-DVLOperations.pdf) should help

New IP of Xavier: 192.168.0.100

chrstrom, 2021-03-14:
The software mentioned in the user manual is on a USB that came with the DVL. I will put it in the top middle locker, if you wish to use it. This is the new configuration:

[image]

chrstrom, 2021-03-14:
Closing this since the DVL IP has been set. For future reference:

To change the IP, you need the .exe from the nortek USB stick. As long as your computer is connected to the same physical network as the DVL, the "Discovery" tab should take care of everything for you :smile:

closed 2021-03-14 by chrstrom