vortexntnu/vortex-auv #203: Mapping      ?

chrstrom, 2021-03-21:
Preferrably we can utilize some soft of plug-and-play mapping software. We have the ESKF as odometry input, but we should consider how the stereo camera point cloud output can be fused with things like sonar data (unsure what format this comes in as of now)

The point cloud from the camera needs to be looked into; how sure do we need to be of features before we make a point out of it?

chrstrom, 2021-03-21:
See issue #154 for other issues that may be causing problems for us here

stefalarse, 2021-04-11:
Goal: find possibilities for mapping, and how far we can go. 

It is possible to achieve a lot of cool things with mapping, but there has not been developed/implemented anything that can be used directly yet. We have both camera and sonar, but "how should we use data from these sensors to understand the local environment around the drone?"

Basically research about what could be worked on with vortex 2022

stefalarse, 2021-04-11:
Short term goal: visualize mapping with zed sdk on Xavier. Deadline April 18th

Long term goal: usable SLAM. Deadline vortex 2022

chrstrom, 2021-04-18:
@stefalarse what's the status for this? The test you made me run on the stormtrooper pc looked very promising 👍

stefalarse, 2021-04-18:
Did not have more time, so that test was the final contribution this semester

chrstrom, 2021-04-21:
Aight, sweet :smile: We have a pretty cool foundation for further development for Vortex 2022!

closed 2021-04-21 by chrstrom