vortexntnu/vortex-auv #206: Getting the Ping360 sonar working

chrstrom, 2021-03-21:
Wohoo! New sonar. See [this issue/repo](https://github.com/CentraleNantesRobotics/ping360_sonar/issues/62#issuecomment-801938068) for a package that may work for serial connectivity with the sonar.

If that proves to be plug-and-play, great 😄 If ethernet connectivity is not hard to program, that is to be preferred.

jae-hyeong-hwang, 2021-04-02:
A bit desperate because of silly error almost similar with this  
https://github.com/CentraleNantesRobotics/ping360_sonar/issues/57#issue-682119965  

I excluded python2.7 in my configuration setting, set default as python3.6. Then I reinstalled everything but still doesn't work.
This is my error
> Traceback (most recent call last):
  File "/home/jaehh/catkin_ws/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
ModuleNotFoundError: No module named 'ping360_sonar'

chrstrom, 2021-04-03:
> 
> 
> A bit desperate because of silly error almost similar with this
> [CentraleNantesRobotics/ping360_sonar#57 (comment)](https://github.com/CentraleNantesRobotics/ping360_sonar/issues/57#issue-682119965)
> 
> I excluded python2.7 in my configuration setting, set default as python3.6. Then I reinstalled everything but still doesn't work.
> This is my error
> 
> > Traceback (most recent call last):
> > File "/home/jaehh/catkin_ws/src/ping360_sonar/nodes/ping360_node", line 3, in 
> > from ping360_sonar import main
> > ModuleNotFoundError: No module named 'ping360_sonar'

Since I got things working without any issues at all, I will try to recreate your issues. We'll stay in touch ye? 😄

chrstrom, 2021-04-10:
From Jae:
> CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
> Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
> which is not found. It does neither exist as an absolute directory nor in
> '${{prefix}}//usr/include/opencv'. Check the issue tracker
> 'https://github.com/ros-perception/vision_opencv/issues' and consider
> creating a ticket if the problem has not been reported yet.
> Call Stack (most recent call first):
> /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
> CMakeLists.txt:10 (find_package)

chrstrom, 2021-04-18:
@theBadMusician @jae-hyeong-hwang Could you put the pool test updates here? 😄 Would love to see something from it!

jae-hyeong-hwang, 2021-04-19:
> @theBadMusician @jae-hyeong-hwang Could you put the pool test updates here? smile Would love to see something from it!

I think @theBadMusician took a screenshot of sonar image via rviz. I also took a photo of it but Benjamin's image might look more clear.

theBadMusician, 2021-04-19:
<ins>Ping360 Sonar Pooltest-4 Summary:</ins>

- The sonar publishes sensible readings using [this ping360 ROS implementation](https://github.com/CentraleNantesRobotics/ping360_sonar) and it looks to be working consistently in general.
- The sonar can be (dynamically and statically) configured in SW according to our needs i.e. range, beam gain, step size, amount of samples, FOV etc.
- The data can be visualized in a B&W top-down-view image and a single-plane point-cloud which can be useful for object detection/avoidance with ROV in low visibility environments (+ it looks really cool).
- Publishes easily understandable topics/data which could be fused with other perception data for use in SLAM next year.
- There is an USB port shuffling issue which happens (sometimes) when the Xavier gets rebooted -- some/most/all Serial devices get re-assigned to different ttyUSB ports (same happens with the powersense Arduino module). Really annoying when it happens. Could be solved by [assigning each device to a static USB port](https://msadowski.github.io/linux-static-port/) or by getting each device's USB port in runtime with a script.
- [This fork](https://github.com/COHRINT/ping360_sonar) expands quite a bit on the functionality of the main ROS repo -- it adds features such as different scan modes, landmark and object detection using logistic regression models and landmark/object tracking. Could be an 'inspiration' for our own sonar functionality expansions in the future.

This issue is being closed, as the sonar is set up and (except for the USB port issue) is working consistently on the Xavier.

closed 2021-04-19 by theBadMusician