vortexntnu/vortex-auv #207: Program OLED display driver

michoy, 2021-03-21:
Electronics has bought an OLED display that can be used to display Xavier's IP address and battery voltage. The display communicates over I2C.

chrstrom, 2021-10-08:
@ingar-eh Could you drop a status report here? I have some notes from the meeting but I think you have a better idea of what's up!

ingar-eh, 2021-11-04:
The only OLED we could find at the office used SPI, not I2C. 

I used my own I2C screen for testing on the Jetson TX2 and managed to display IP. Shortly after this success I managed to fry my I2C so we currently have no I2C screen and I've postponed reproducing on the Xavier until we have a screen. Displaying battery voltage shouldn't be an issue after Issue #276 is resolved.

closed 2022-01-13 by chrstrom