vortexntnu/vortex-auv #221: Temperature measurement crash

michoy, 2021-04-04:
The temperature monitor crashes when run on the Xavier

cat: /sys/kernel/debug/bpmp/debug/soctherm/group_CPU/temp: Permission denied
Traceback (most recent call last):
  File "/home/vortex/vortex_ws/src/Vortex-AUV/mission/internal_status/src/temperature_monitor.py", line 44, in <module>
    tm.spin()
  File "/home/vortex/vortex_ws/src/Vortex-AUV/mission/internal_status/src/temperature_monitor.py", line 29, in spin
    self.measure_temp()
  File "/home/vortex/vortex_ws/src/Vortex-AUV/mission/internal_status/src/temperature_monitor.py", line 38, in measure_temp
    self.cpu_temperature = int(subprocess.check_output(["cat", self.path_to_cpu_temperature_meter]).decode("utf-8"))
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['cat', '/sys/kernel/debug/bpmp/debug/soctherm/group_CPU/temp']' returned non-zero exit status 1

Areskiko, 2021-04-11:
Do we know if the error occurs immediately or only after some time?

Areskiko, 2021-04-11:
It may be possible to use [tegrastats](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3243/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2FAppendixTegraStats.html%23wwpID0E4HA)

michoy, 2021-04-11:
It appears immediately after launch

chrstrom, 2021-04-18:
Can this be closed with the merge of pull #225?

Areskiko, 2021-04-18:
I assume so

chrstrom, 2021-04-21:
nice ^

closed 2021-04-21 by chrstrom