vortexntnu/vortex-auv #262: ORB-SLAM3 with own camera feed

chrstrom, 2021-10-05:
**Estimated time:** 
**Deadline:** 10. october

**Description:** ORB-SLAM3 has been ran using an example video file, but we want to run it with our own camera feed from the ZED camera 😄

crose72, 2024-03-26:
Hi @chrstrom, I am using an NVDIA Jetson Nano and a CSI ribbon camera (regular monocular and stereo inertial) and I'm having some trouble getting ORB SLAM3 to work with my live camera feed.  The issue I have is that I can't enable the visualizer to see how SLAM is working on that video footage.  I got the EuRoC example with the visualizer enabled working.  My code will run, but it seems that the video feed isn't being passed to the SLAM system correctly and therefore the map isn't being created.  Do you think you could help me getting this code up and running?  The internet really only shows off what SLAM is and doesn't have many resources to help you get started with your own camera lol.

martongonczy, 2024-06-26:
Hi @crose72,

I have the same issue. Have you been able to get it to work?

crose72, 2024-06-28:
> Hi @crose72,
> 
> I have the same issue. Have you been able to get it to work?

Yes, I am at least now actually seeing the video frames in the viewer and the mapping/localization is working.  You wanna know what I changed in my code?  Nothing lol.  I just switched to a different jetson nano and then it worked all of a sudden.  Try this code https://github.com/crose72/OperationSquirrel/tree/master/scratch/SLAM-examples.  Specifically the `live-slam-example` and `live-slam-example-2`.  Takes a second to load up, but it's now working out of the box for me.

martongonczy, 2024-06-28:
> > Hi @crose72,
> > I have the same issue. Have you been able to get it to work?
> 
> Yes, I am at least now actually seeing the video frames in the viewer and the mapping/localization is working. You wanna know what I changed in my code? Nothing lol. I just switched to a different jetson nano and then it worked all of a sudden. Try this code https://github.com/crose72/OperationSquirrel/tree/master/scratch/SLAM-examples. Specifically the `live-slam-example` and `live-slam-example-2`. Takes a second to load up, but it's now working out of the box for me. I didn't calibrate the camera yet though, I just wanted to see the slam working so I'm using a borrowed yaml file.

Thank you for your help. I will check it out. It started working for me as well, it seems like that the visualiser needs some time to start also I need to start moving the camera around otherwise the mapping does not start on the visualiser.