vortexntnu/vortex-auv #272: Debug /guidance/move

finnfi, 2021-10-16:
**Time estimate:**  10 h
**Deadline:**  24.10.2021
**Teams SCRUM link:** [Insert link here ->]()  

**Description of task:** Get the fsm we have back on track. Test dp_guidance and los_guidance.

finnfi, 2021-10-16:
Changed so that fsm's are working again. Have to change the logic /guidance/move action server to fsm folders. Have to debug some warnings while running fsms. Currently only testet four corner mission. Have to make a param that can choose right output from thruster_merger for simulator vs real drone. Currently publishing to both.

TarekEl-Agroudi, 2021-10-17:
Simplified guidance interface. Previously, both dp_move() and los_move() functions called a sepparate action server (guidance/move) which in turn called the dp/los action server. This was highly redundant, so the intermediate action server level is now removed, so that dp_move() directly calls the dp action server, and los_move() directly calls the los action server.

Remaining work: some error messages when calling fsm's, generally in the form "[ERROR][...] [guidance/interface]: Recieved XXX when in state XXX

TarekEl-Agroudi, 2021-10-31:
Branch bug/DP_move was tested on Beluga. The error messages from the comment above did show up, but these are not a critical error. However, a new error message came up: Attempt to get a goal id on uninitialised ServerGoalHandle (/in guidance/dp)

closed 2022-01-12 by finnfi