vortexntnu/vortex-auv #283: Calibrate ZED camera with casing both above and underwater

ivangushkov, 2021-11-02:
**Time estimate:** 10 hours  
**Deadline:** 12.11

**Description of task:**
Camera needs to be calibrated. It probably has the factory calibration, which does not take the casing into account. This should be maybe be done by one from CV and one from SLAM.

KristianSem, 2021-11-02:
[How do I recalibrate my ZED stereo camera](https://support.stereolabs.com/hc/en-us/articles/360011828773-How-do-I-recalibrate-my-ZED-stereo-camera-) - Stereolabs

KristianSem, 2021-11-03:
We should calibrate the camera for both underwater and overwater so that we are able to switch between the values as needed + might get an idea on how water affects the different parameters

KristianSem, 2021-11-08:
After some testing last night we figured that using [Camera_calibration](http://wiki.ros.org/camera_calibration) from ROS will work. Running:

`rosrun camera_calibration cameracalibrator.py --approximate 0.1 --size 8x6 --square 0.108 right:=/zed2/zed_node/right_raw/image_raw_color left:=/zed2/zed_node/left_raw/image_raw_color --no-service-check`

With the camera_calibration package it seemed to work (NOTE: The number of square and square size was off, this will be done properly during the true calibration). 

The output will be something like this:

<pre>
[image]

width
672

height
376

[narrow_stereo/left]

camera matrix
262.669355 0.000000 333.280105
0.000000 263.899997 189.779156
0.000000 0.000000 1.000000

distortion
-0.027536 -0.003140 0.000275 -0.000416 0.000000

rectification
0.999949 -0.003041 0.009595
0.003011 0.999990 0.003222
-0.009605 -0.003193 0.999949

projection
263.069694 0.000000 326.263634 0.000000
0.000000 263.069694 191.028177 0.000000
0.000000 0.000000 1.000000 0.000000

[image]

width
672

height
376

[narrow_stereo/right]

camera matrix
261.683948 0.000000 339.943399
0.000000 263.405154 190.483845
0.000000 0.000000 1.000000

distortion
-0.033662 -0.000510 0.000948 -0.002391 0.000000

rectification
0.999943 -0.002554 0.010334
0.002587 0.999992 -0.003194
-0.010326 0.003221 0.999941

projection
263.069694 0.000000 326.263634 -93.556963
0.000000 263.069694 191.028177 0.000000
0.000000 0.000000 1.000000 0.000000
</pre>

But we want it on this format and how to it is explained on OpenCV: [camera_calibration_and_3d_reconstruction](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html)
<pre>
[LEFT_CAM_VGA]
fx=263.8025
fy=263.6775
cx=333.3125
cy=185.438
k1=-0.0408546
k2=0.00958227
k3=-0.00489882
p1=-0.000394428
p2=0.000536681
</pre>

Remember to read about how the parameters can be scaled between resolutions (e.g. 1920/1080 = scalar)

KristianSem, 2021-11-08:
Work in progress tasks:
- [ Make a complete guide. Installing software and calibration procedure]
- [Get a checkerboard] 

[Example of checkerboard](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration?action=AttachFile&do=view&target=check-108.pdf)

SimonVolden, 2021-11-14:
Dear SLAM/CV
Your comment is wrong it should actually be:
`rosrun camera_calibration cameracalibrator.py --approximate 0.1 --size 8x6 --square 0.108 right:=/zed2/zed_node/right_raw/image_raw_color left:=/zed2/zed_node/left_raw/image_raw_color left_camera:=/zed2/zed_node/left right_camera:=/zed2/zed_node/right --no-service-check`

easy peasy lemoin squeezy

KristianSem, 2021-11-14:
How to get the calibration tool
Clone the image_pipeline repo:
`$ git clone https://github.com/ros-perception/image_pipeline.git`
`$ cd image_pipeline`
`$ git checkout melodic`

Then make a calibration workspace and copy the camera_calibration into the src folder:
`$ cd ~/`
`$ mkdir -p calib_ws/src`
`$ cp -r ~/image_pipeline/camera_calibration ~/calib_ws/src/`
`$ cd calib_ws`
`$ catkin build`

Bruglen, 2021-11-14:
We have decided to get a checkerboard for underwater calibration, can't get much more progress until then.