vortexntnu/vortex-auv #286: Create a new fsm for the pre-qualification task

finnfi, 2021-11-07:
**Time estimate:** 30 hours  
**Deadline:**  24.12.2021

**Description of task:** Create a first implentation of the fsm for the prequalifing task. First make it work in the simulator, before testing on the actual drone. This can in the end be a base for the final fsm.

TarekEl-Agroudi, 2022-01-10:
Preliminary state machine works:
- Landmarks server up and running. Publishing a gate position on topic "/fsm/object_positions_in" will update the object_positions map
- Simple state machine works and waits for object position, and once it arrives succsesfully moves the drone there.

Plan for the week:

- Implement remaining states for prequalification task. 
- When it is ready, replace dp/los with virtual target path following (where suitable)
- Integrate and communicate with CV/OD/SLAM groups

TarekEl-Agroudi, 2022-01-11:
Pole FSM up and running as well

TarekEl-Agroudi, 2022-01-12:
NB: Must change from using simple action states (e.g. dp_move ) to using actions directly. This is so that we can update the goal position during the converge state

TarekEl-Agroudi, 2022-01-14:
Update: Started integration of virtual target path following to fsm and looks promising. Can update target in runtime by simulating CV inputs, and path response accordingly. Minor bugs:

- sometimes, the vtf action server finishes prematurely
- we need to manually stop the dp controller during the reach_depth state by setting control mode to open loop. There should be a better way of doing this

TarekEl-Agroudi, 2022-01-17:
Update: The FSM is complete when it comes to states and functionality. It finds the gate, moves in front of it, moves through it, finds the pole, moves in front of it, moves around it in a circle, finds the gate again and returns through it...

The biggest issue seems to be highly aggresive VTF tuning or too fast virtual target speed. Hopefully just a matter of tuning.

closed 2022-01-18 by TarekEl-Agroudi