vortexntnu/vortex-auv #303: Document and refacture the published odometry frames from the robot localization

theBadMusician, 2022-01-25:
**Time estimate:** 15 hours  
**Deadline:**  01.02.22

**Description of task:**
The odometry from the robot localization EKF is published in ENU frame. This needs to be changed to NED frame and properly documented. Go over the inputs into the EKF from the STIM300 and DVL1000, and make sure that everything matches. The topic names regarding odometry also should be refactored to include its frame: "/odometry/filtered" -> "/odometry/filtered/enu".

TarekEl-Agroudi, 2022-02-06:
Comment: TFs to ENU from sensors now in launch file for ekf (correct version, hopefully) AND in beluga.launch (most likely incorrect there). Should have them one place, ideally in a sepparate launch file dedicated to TFs.

chrstrom, 2022-04-26:
Closing this since this work is just about done. The EKF performs very very well now!

closed 2022-04-26 by chrstrom