vortexntnu/vortex-auv #323: Remake vortex_msgs to its own repo again?

chrstrom, 2022-05-10:
**Time estimate:** 0 hrs
**Deadline:** NA

**Description of task:** vortex_msgs used to be its separate repo, but was moved into here after realizing that no other repo really uses the messages. Now that we have a proper structure in place, we are once again using the messages across multiple repo. While "you will likely never use another repo while *not* also using Vortex-AUV" I still think this could be a healthy update.

Do note that this will reintroduce the trouble of synchronizing changes across multiple repos, but that is realistically not that big of an issue now.

@theBadMusician @KristianSem thoughts?

theBadMusician, 2022-05-10:
Definitely a needed change :+1: We've encountered quite a bit if annoyance when working on both the [Vortex-CV](https://github.com/vortexntnu/Vortex-CV) and the [Tritech Sonar](https://github.com/vortexntnu/720i-sonar-linux-driver) repos, and having to manually move the vortex_msgs pkg into appropriate workspaces. There are enough non-Vortex-AUV packages, which are dependent on vortex_msgs, to make it a self-contained repo again.

theBadMusician, 2022-09-24:
#334 resolves this issue.

closed 2022-09-24 by chrstrom