vortexntnu/vortex-auv #344: Implement autonomous valve turning functionality

alekskl01, 2022-11-20:
**Time estimate:** N/A  
**Deadline:**  N/A

**Description of task:** Add functionality for use of the gripper to turn valves for use in the TAC competition

Lasgut, 2023-01-13:
The task will be performed by usage of an smach state machine divided in three parts.
1. ValveSearch: the drone will be controlled manually until the valve is visual on the camera, therefore this state will be used to confirm that Beluga have a visual of the valve.
2. ValveConverge: it is important that the drone's alignment and orientation is maching the valve, therefore this part will include a regulation in terms of alignment and orientation and if this is within a defined acceptance the drone shall converge towards the valve.
3. ValveExecute: This shall include a confirmation that the valve is inside the gripper, then figure out which direction the valve shall be turned and then execute the rotation before extract from the valve.

Main issues that should be addressed:
- what type of controller that should be used, DP or VTF
- limitations and possibilities in terms of perception

Lasgut, 2023-01-13:
The 1. state will not be present in this task. The confirmation of a visual of the valve will be done in the state machine changing from manual operation to the task. 
A more detailed implementation plan is made in valve_horisontal.py

Lasgut, 2023-01-22:
The gripper position is found: x = -1.98mm, y = -171mm, z = -259.56mm, with x as forward and z upwards. Further the interface with the gripper is updated such that the joystick interface topic is used. We will use the DP controller to converge to the valve therefore the message which will be used for this purpose will be updated when its ready. The next part will be to test the code and publish on different topics. 

It will probably be difficult to make this task fully autonomous therefore a possible solution is to combine autonomous motion with decision making from an operator looking at live camera feed.

chrstrom, 2023-01-22:
> 

Just a note: X forward and Z upwards constitute a North-East-Up (NEU) right-hand system, which afaik is not used in the auv stack. For reference, we prefer working in NED, but robot_localization expects IMU measurements to be in ENU, and produces estimates in ENU. I am unsure what the status is for each package that depends on coordinate convention, but in any case it is something to address (in a separate issue likely).

The problem for this task could arise in differences in coordinate conventions making the DP controller converge to the wrong point😢

closed 2024-02-26 by alekskl01