vortexntnu/vortex-auv #345: Add framework for corntrol of autonomous states during manual operations

alekskl01, 2022-11-20:
**Time estimate:** N/A 
**Deadline:**  End of arbeidsuke

**Description of task:** Implementation of a framework allowing us to switch to an autonomous state from manual control for completion of tasks.

Sigbra, 2023-01-10:
I will start to implement the state "manual mode" in the tac_fsm.

Sigbra, 2023-01-13:
Will be using dynamic_reconfigure to set up the state machine in a way that makes it possible to switch between a manual and autonomous modes. A autonomous mode represent a task. Have created a CME package and written a config file.

Sigbra, 2023-01-13:
Dynamic_reconfigure implementation finished. More states can be added later. When testing, CME server and client node must run before using "rosrun rqt_reconfigure rqt_reconfigure Tac_States".

Sigbra, 2023-01-22:
Code is a mess right now because of try and fail coding. It runs, but it does not do what it is supposed to (which is change between task_1_node and task_2_node with the dynamic_reconfigure").

closed 2023-03-12 by Sigbra