vortexntnu/vortex-auv #352: Update function for getting PSM method from new PDB

PizzaAllTheWay, 2023-01-22:


PizzaAllTheWay, 2023-01-29:
ADC register for I2C: 0x69

Tips: Use this repository: https://github.com/stevemarple/python-MCP342x/blob/master/MCP342x/examples/convert_and_read_many.py

Use this library: https://pypi.org/project/MCP342x/

PizzaAllTheWay, 2023-02-02:
#!/usr/bin/env python
import glob
import logging
import smbus
import time
from MCP342x import MCP342x
convertionRatio = 1.636579823702253 
ofset = 0.04321161085945153
bus = smbus.SMBus(1)
ch0 = MCP342x(bus, 0x69, channel=0, resolution=18)
ch1 = MCP342x (bus, 0x69, channel=1, resolution=18)
i=0
while True:
i += 1
voltage = round(ch0.convert_and_read() * convertionRatio ofset, 3)
current = round(ch1.convert_and_read() * convertionRatio ofset, 3)
print(
f"Voltage: (voltage} \t|     Current: (current} \t|     update: {i}"
)

PizzaAllTheWay, 2023-02-02:
Use the code above jesjes

PizzaAllTheWay, 2023-02-05:
Edit the launch file when done: https://github.com/vortexntnu/vortex-auv/blob/development/mission/internal_status/launch/internal_status.launch

chrstrom, 2023-02-05:
Does the existing code here
 https://github.com/vortexntnu/vortex-auv/blob/development/mission/internal_status/src/battery_monitor.py

not work for this?

PizzaAllTheWay, 2023-02-05:
We can try it out first jesjes 👍

PizzaAllTheWay, 2023-02-12:
We are almost done, testing left :)

PizzaAllTheWay, 2023-02-13:
- self.i2c_address should be set with rosparam()
- adc_convertion_ratio and adc_offset should not be an empirical value, it should be stated somewhere in the datasheet and set to exact values from here: https://www.microchip.com/en-us/product/MCP3422

PizzaAllTheWay, 2023-02-13:
Updated code. Now we just need to test it out jesjes :)

PizzaAllTheWay, 2023-02-17:
Done testing, some changes made, everything good now
Documentation on stuff on wiki coming soon .tm here: 
https://vortex.a2hosted.com/index.php/PSM
and here:
https://vortex.a2hosted.com/index.php/Beluga_PDB

closed 2023-02-19 by TheoHHH