vortexntnu/vortex-auv #357: cant pull image from ghcr

FabianKongelf, 2023-01-27:
"når bruker kommando docker-compose up: Head "https://ghcr.io/v2/vortexntnu/vortex-auv/manifests/development": unauthorized" - ronja.

There seems to be an issue when pulling docker image from ghcr related to permissions, is this uniq to ronja? or do the issue reoccur with others?

FabianKongelf, 2023-01-27:
https://github.com/docker/hub-feedback/issues/1484

chrstrom, 2023-01-27:
Likely an expired access token that is cached. Reset by docker login ghcr.io/vortexntnu

FabianKongelf, 2023-02-20:
Don't know, though it was solved, is it not? I think @theBadMusician changed some containers to open access (from private) so access rights should not matter, is this done to this repo / can it be done?

theBadMusician, 2023-02-20:
Closing, as the pkg is now public. There shouldn't be any authorization issues. 

The issue can be reopened if something encounters the issue again.

closed 2023-02-20 by theBadMusician