vortexntnu/vortex-auv #359: IIR filter for OLED

PizzaAllTheWay, 2023-01-29:


chrstrom, 2023-01-29:
We concluded that this is not the only source of error. The voltage measurements went up to 23-25V when running the thrusters. In all likelihood this is because of a fluctuation in the input voltage to the arduino that does the A to D conversion. A lower reference voltage than expected will yield a higher converted voltage (erroneously). Consult with Henrik about this when the elhouse is reworked.

chrstrom, 2023-01-29:
Additionally there seem to be inconsistencies between the OLED and the ros topic that published voltage, so maybe look into that 😄

PizzaAllTheWay, 2023-02-17:
Done on OLED screen, just change libraries to LCD screen and will work for LCD as well :)

closed 2023-02-17 by PizzaAllTheWay