vortexntnu/vortex-auv #432: [TASK] Expand joystick interface to have a mode that publishes a guidance reference message

Andeshog, 2024-09-16:
### Description of task

Right now the joystick interface publishes a wrench message. In practice, if there are strong currents in the ocean it is more difficult for a human operator to control the drone. To ensure better control, a mode that publishes a reference signal instead of a control input should be implemented. Unnecessary/repetetive logging should also be removed.

### Suggested Workflow

- Get familiar with how the current [joystick interface](https://github.com/vortexntnu/vortex-auv/blob/development/mission/joystick_interface_auv/joystick_interface_auv/joystick_interface_auv.py) works by testing it out in the simulator.
- Get familiar with the NED coordinate system and the 6 degrees of freedom for an underwater vehicle.
- Write the code and test it by plotting all degrees of freedom in plotjuggler or foxglove.

### Specifications

- Use [Float32MultiArray](https://docs.ros.org/en/jade/api/std_msgs/html/msg/Float32MultiArray.html) as the message type for the reference signal. It should be filled with 6 values (pos_north, pos_east, pos_z, roll_angle, pitch_angle, yaw_angle).
- Use "joystick/guidance" as the topic name.
- The starting pose (position and orientation) for the mode should be the current pose of the drone, but when developing the code feel free to start at [0 0 0 0 0 0].

### Contacts

- @Andeshog

### Code Quality

- [ ] Every function in header files are documented (inputs/returns/exceptions)
- [ ] The project has automated tests that cover MOST of the functions and branches in functions (pytest/gtest)
- [ ] The code is documented on the wiki (provide link)

chrstrom, 2024-09-18:
@Andeshog you might want to consider using a geometry_msgs::msg::PoseStamped for this (or some other well-suited standard message with a header). With a generic array you require the "user" to both know what index corresponds to which state and to correctly parse it. A Pose message will make that explicit (plus you can use all the tf2/Eigen conversion functions), and having the timestamp field populated is always a good idea (you will thank yourself later for timestamping your messages). Having the header also means you get to explicitly define the frame your data is in.

Note that nested messages are a bit EVIL in terms of performance (especially for serialization/deserialization), but that's honestly not a concern with such little data being sent.

Andeshog, 2024-09-18:
> @Andeshog you might want to consider using a geometry_msgs::msg::PoseStamped for this (or some other well-suited standard message with a header). With a generic array you require the "user" to both [...]

Yes, totally agree. Thanks for pointing out

closed 2025-01-30 by Andeshog