vortexntnu/vortex-auv #439: [TASK] Implement EKF for sensor fusion of Nucleus and IMU

Andeshog, 2024-09-19:
### Description of task

To get better pose and velocity estimates, we can fuse the Nucleus measurements and the IMU measurements. The measurements from the Nucleus are given in an [INS](https://github.com/NortekSupport/nucleus_driver/blob/main/ros2/src/interfaces/msg/INS.msg) message and the IMU measurements are given in an [IMU](https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html) message.

The IMU can be tested in the office by:
- Finding the IMU and plugging it into your computer.
- Make a new workspace, clone [this repo](https://github.com/vortexntnu/stim300-driver/tree/feature/ros2-port) and build:
    - `mkdir -p stim300_ws/src ; cd stim300_ws/src` 
    - `git clone git@github.com:vortexntnu/stim300-driver.git -b feature/ros2-port`
    - `cd .. ; colcon build ; source install/setup.bash`
- Launch the driver:
    - `ros2 launch driver_stim300 stim300_driver.launch.py`

### Suggested Workflow

- Define the state vector to be estimated. Must include 6 DOF pose (NED) and velocity (body).
- Run and read the output from the IMU by doing the steps in the description.
- Look at [this](https://github.com/vortexntnu/vortex-vkf/blob/main/vortex-filtering/include/vortex_filtering/filters/ekf.hpp) template for inspiration/starting point.

### Specifications

- Output message type: [Odometry](https://docs.ros2.org/foxy/api/nav_msgs/msg/Odometry.html)
- Output topic name: "nucleus/odom"

### Contacts

- @Andeshog 
- @chrstrom

### Code Quality

- [ ] Every function in header files are documented (inputs/returns/exceptions)
- [ ] The project has automated tests that cover MOST of the functions and branches in functions (pytest/gtest)
- [ ] The code is documented on the wiki (provide link)

Talhanc, 2024-09-22:
We are thinking of implementing the error state kalman filter (ESKF). We will leave this for now, and try to learn c++ more, and learning the ESKF from the sensorfusjon course, and then trying to implement it in late january/february. Hopefully until then we have more info and structured how we want to program it,

Talhanc, 2025-02-09:
Just pushed the current python version, is not what i wanted to have but will do for the moment. Need to setup the conversions from msgs to array, and do a test. Hopefully some time next week