vortexntnu/vortex-auv #485: [TASK] AUV GUI

Hallfred, 2024-10-13:
### Description of task

Make a GUI that plots odometry data and can be expanded with more functionality

### Code Quality

- [ ] Every function in header files are documented (inputs/returns/exceptions)
- [ ] The project has automated tests that cover MOST of the functions and branches in functions (pytest/gtest)
- [ ] The code is documented on the wiki (provide link)

Andeshog, 2024-11-12:
I think it would be nice to look into showing the GUI in the browser, by using Flask for example. Are you able to research this @Hallfred? If there are clear benefits to do it this way, then I suggest we try it out. I would image chat being able generate some nice templates to get us started.

Hallfred, 2024-11-12:
> I think it would be nice to look into showing the GUI in the browser, by using Flask for example. Are you able to research this @Hallfred? If there are clear benefits to do it this way, then I suggest we try it out. I would image chat being able generate some nice templates to get us started.

Can probably start reading up on it, but the rest of the semester is pretty busy for my part.

Andeshog, 2024-11-12:
> Can probably start reading up on it, but the rest of the semester is pretty busy for my part.

Great! No need to do it this semester, you can start looking at it in the workshop week 👍

kluge7, 2024-11-18:
Since the GUI doesn't need to run on Orca, it should in the future be moved to this repository: https://github.com/vortexntnu/vortex-visualization

kluge7, 2024-11-18:
Please add a requirements.txt file for dependencies not supported by rosdep (e.g., PyQt6).

Hallfred, 2025-01-06:
The GUI code has been copied to [here](https://github.com/vortexntnu/vortex-visualization/tree/auv_gui), however I know that #502 uses part of the GUI package. Not sure how to move forward with removing the code from this repo

Andeshog, 2025-01-06:
That pull request should not use the GUI package, it was just done for testing purposes. So it can safely be removed from the branch

closed 2025-02-23 by kluge7