vortexntnu/vortex-auv #577: [Task] ESKF fuse depth sensor

Q3rkses, 2025-09-20:


jorgenfj, 2025-09-22:
> Discussed this with [@Talhanc](https://github.com/Talhanc), and it might be in order to update the ESKF code with a sensor concept (for anything but the IMU). Something like
> 
> template<typename T>
> concept SensorModelConcept = requires(const T &meas, const NominalState &state)
> {
>     { meas.innovation(state) } -> std::convertible_to<Eigen::VectorXd>;
>     { meas.jacobian(state) } -> std::convertible_to<Eigen::MatrixXd>;
>     { meas.noise_covariance() } -> std::convertible_to<Eigen::MatrixXd>;
> };
> 
> with a corresponding method in the ESKF
> 
> template<SensorModelConcept SensorModel>
> void update(const SensorModel& sensor) {}
> 
> For any new sensor you would then just implement
> 
> struct MyNewCoolAndFancySensor {
>     Eigen::Vector3d measurement;
>     Eigen::Matrix3d measurement_noise;
>     Eigen::VectorXd innovation(const NominalState &state) const {}
>     Eigen::MatrixXd jacobian(const NominalState &state) const { }
>     Eigen::MatrixXd noise_covariance() const {}
> };
> 
> and a corresponding ros-translator
> 
> Just a way to simplify adding new sensors to the filter 💯

^^ This is now a requirement from the board

AhmedBorch, 2025-10-05:
Hi:), should I use the new branch instead of feeterror-state-kalman-filter? @Talhanc the NIS explodes everytime I test the eskf, do you know whether the DVL covariance could be the reason? Also, why does the IMU frame has that rotation (in eskf_params.yaml)?

Talhanc, 2025-10-06:
I will send you a teams message :) @AhmedBorch